berlin-open-wireless-lab/DAWN #233: [Info] Updates to luci-app-dawn in progress

danielvijge, 2023-10-27:
As a notification, I submitted merge request https://github.com/openwrt/luci/pull/6656 to convert `luci-app-dawn` from lua to JavaScript.
Besides the conversion, it also includes some improvements:

- In both the network overview and the hearing map MAC addresses are replaced by host name if known.
- In the hearing map, the table is sortable. If the same client is connected to multiple access points/frequencies the MAC/host name is listed twice, where in the lua implementation the second MAC address was empty to show it was referring to the same client.
- Aligning the column names between the network overview and the hearing table.
- Add tooltips for abbreviations.

There are some things that might be a bit weird, but they are coming from the ubus data:
- The RCPI (Received Channel Power Indication) and RSNI (Received Signal to Noise Indicator) always display as -1.
- The JSON feed isn't that nice to work with, as it doesn't return an array, but only named objects.

Are there might requests to add? Then I might see if I can add them too.

danielvijge, 2023-10-29:
For those who want to test it, here is a build. Officially this is a build for mips_24kc, but it might work on other targets too, as it contains no binary code.
Unzip it, then install it using `opkg install luci-app-dawn*.ipk`.

**Important**:  You might see an error after a login. This is more a limitation of the package manager, than an error in this new version. If the old `luci-app-dawn` was the only package that needed `luci-compat` this package will be removed. But the packages needed by `luci-compat` will not be removed if they are no longer needed. To fix this either:
- (safest option) reinstall `luci-compat` with `opkg install luci-compat`.
- (possibly breaking option) if you know the lua compatibility layer isn't used by any other package, remove everything with ``opkg remove luci-compat luci-lua-runtime ucode-mod-lua luci-lib-base lua`.

[luci-app-dawn_JS-version.zip](https://github.com/berlin-open-wireless-lab/DAWN/files/13198984/luci-app-dawn_JS-version.zip)

PolynomialDivision, 2023-10-31:
It would be awesome to also include:
https://github.com/blogic/ufp

PolynomialDivision, 2023-10-31:
Can you make sure that you see them in the terminal calling `ubus call dawn get_network`?

XiaoliChan, 2023-10-31:
> Can you make sure that you see them in the terminal calling `ubus call dawn get_network`?

Only one AP data
[image]

PolynomialDivision, 2023-10-31:
Can you check `ubus call umdns browse`?

XiaoliChan, 2023-10-31:
> Can you check `ubus call umdns browse`?

Got blank result
[image]

PolynomialDivision, 2023-10-31:
@XiaoliChan That's a umdns issue and not dawn related. Please check umdns config and so on. :)

XiaoliChan, 2023-11-01:
> @XiaoliChan That's a umdns issue and not dawn related. Please check umdns config and so on. :)

Wired, I didn't touch the umdns stuff, I just update my firmware to latest version, after that, Dawn not working anymore

XiaoliChan, 2023-11-01:
Here is the screenshot 
[image]

XiaoliChan, 2023-11-01:
@PolynomialDivision Follow the issues #56, it solved my issues!

PolynomialDivision, 2023-11-01:
So you installed `procd-seccomp`?

XiaoliChan, 2023-11-01:
> So you installed `procd-seccomp`?

I have already installed it, but I just execute these commands
```
cd /etc/seccomp
mv umdns.json umdns1.json
sync
/etc/init.d/umdns restart 
```

danielvijge, 2023-11-01:
> It would be awesome to also include: https://github.com/blogic/ufp

Should be doable as soon as a package is released. It exposes a feed, which is not different than the `hostHints`. Only question might be if `ufp` should become a dependency of `luci-app-dawn`. If it's optional, it might be a bit more complex to test if it is installed or not.

danielvijge, 2023-11-02:
Closing this info issue, as the updated app is released by a merge in [main](https://github.com/openwrt/luci/commit/ea8c0aa2a1a1ff9f29ffd452049a7b749298ed17) and [openwrt-23.05](https://github.com/openwrt/luci/commit/2eb74226a43e407e03edcf43ac62269196f3e7de)

closed 2023-11-02 by danielvijge