berlin-open-wireless-lab/DAWN #237: DAWN does not see other APs until umdns is restarted

lukasubo, 2024-01-01:
Model | Linksys E8450 (UBI)
Architecture | ARMv8 Processor rev 4
Target Platform | mediatek/mt7622
Firmware Version | OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-23.306.39416-c86c256

Hi, I am having an issue where after booting my APs, DAWN does not see any other APs until I manually restart the umdns service. Once I restart it on all APs, the network overview and hearing map are populated immediately.

rany2, 2024-01-06:
It's a umdns bug not an issue with DAWN. If you do `ubus call umdns browse` you'll notice that the nodes aren't present, upon restart they appear again. I'm not sure what the cause of this is but I think it's a relatively new bug with umdns.

cyyself, 2024-01-19:
same issue here with OpenWrt 23.05.2.

NilsRo, 2024-02-29:
Any workaround available and issue opened in umdns? Put umdns in cron for restarting but would be nice to get it fixed. Using also 23.05.2.

BigMaxTs, 2024-03-05:
same story with the OpenWrt 23.05.0 r23497-6637af95aa / LuCI openwrt-23.05 branch git-23.236.53405-fc638c8 on the Linksys EA8500

teeedubb, 2025-02-13:
I see that umdns has received an update in 24.10:

2024.09.17~fbaca4b6-r1 » 2025.02.10~c5ca22a7-r1

I will upgrade on both routers and report back if this fixes DAWN of 24.10

Z10N0110, 2025-02-14:
Ran into the same problem.  

model `Nokia EA0326GMP`
op version `OpenWrt 24.10.0 r28427-6df0e3d02a`
umdns version `2025.02.10~c5ca22a7-r1`

so i can confirm that is NOT fixed.

hexoxssaa, 2025-03-14:
> Ran into the same problem.
> 
> model `Nokia EA0326GMP` op version `OpenWrt 24.10.0 r28427-6df0e3d02a` umdns version `2025.02.10~c5ca22a7-r1`
> 
> so i can confirm that is NOT fixed.

It could be fixed by rename the umdns.json file to disable seccomp on 23.05.
Now it doesn't work anymore on 24.10.
That's very frustrating

hexoxssaa, 2025-03-14:
> Ran into the same problem.
> 
> model `Nokia EA0326GMP` op version `OpenWrt 24.10.0 r28427-6df0e3d02a` umdns version `2025.02.10~c5ca22a7-r1`
> 
> so i can confirm that is NOT fixed.

mv the /etc/seccomp/umdns.json to /etc/seccomp/umdns.json.disable may work