berlin-open-wireless-lab/DAWN #238: BEACON REQUEST failed

Pablo1732, 2024-01-10:
Hello,
I've been trying to get dawn up and running for a month now. I have already read through several forum posts, the OpenWRT Wiki and the Install.md/Configure.md on GitHub. However, Dawn has not worked properly once so far, the devices decide for themselves when they are roaming not dawn. I've even reset my router and APs and set everything up again.
I am using OpenWRT 23.05.2.

The error "BEACON REQUEST failed" always appears in the console for each of my devices.

(Here is a short excerpt of the log)

Wed Jan 10 16:04:21 2024 daemon.warn dawn: Client / BSSID = ZZ:ZZ:ZZ:ZZ:ZZ:ZZ / YY:YY:YY:YY:YY:YY: BEACON REQUEST failed
Wed Jan 10 16:04:21 2024 daemon.warn dawn: Client / BSSID = ZZ:ZZ:ZZ:ZZ:ZZ:ZZ / YY:YY:YY:YY:YY:YY: BEACON REQUEST failed
Wed Jan 10 16:04:21 2024 daemon.warn dawn: Client / BSSID = XX:XX:XX:XX:XX:XX / YY:YY:YY:YY:YY:YY: BEACON REQUEST failed
Wed Jan 10 16:04:34 2024 daemon.warn dawn: Client / BSSID = ZZ:ZZ:ZZ:ZZ:ZZ:ZZ / YY:YY:YY:YY:YY:YY: BEACON REQUEST failed
Wed Jan 10 16:04:34 2024 daemon.warn dawn: Client / BSSID = ZZ:ZZ:ZZ:ZZ:ZZ:ZZ / YY:YY:YY:YY:YY:YY: BEACON REQUEST failed
Wed Jan 10 16:04:34 2024 daemon.warn dawn: Client / BSSID = XX:XX:XX:XX:XX:XX / YY:YY:YY:YY:YY:YY: BEACON REQUEST failed
Wed Jan 10 16:04:41 2024 daemon.warn dawn: Client / BSSID = ZZ:ZZ:ZZ:ZZ:ZZ:ZZ / YY:YY:YY:YY:YY:YY: BEACON REQUEST failed
Wed Jan 10 16:04:41 2024 daemon.warn dawn: Client / BSSID = XX:XX:XX:XX:XX:XX / YY:YY:YY:YY:YY:YY: BEACON REQUEST failed

(Examples of which pages I have used)
https://forum.openwrt.org/t/dawn-a-decentralized-wireless-controller/108768/224
https://openwrt.org/docs/guide-user/network/wifi/dawn
https://github.com/berlin-open-wireless-lab/DAWN/blob/master/CONFIGURE.md
https://github.com/berlin-open-wireless-lab/DAWN/blob/master/INSTALL.md
https://forum.openwrt.org/t/finally-got-802-11v-working-fully-few-pointers/56678/93

lars18th, 2024-04-23:
Hi @Pablo1732 ,

I've a similar problem. I configured Dawn about two years ago using it on two APs in the same domain. And these messages keep popping up. I'm not sure of the source of the error.

lars18th, 2025-04-22:
Hi @Pablo1732 (and others),

The root cause is now identified: https://github.com/openwrt/packages/issues/26280