berlin-open-wireless-lab/DAWN #245: How often does DAWN replicate?

Darin755, 2024-12-13:
How often does DAWN share its data with other peers? Is this a controllable setting? I know that there are trade offs no matter how you implement the fanout and I am curious about how it is done in dawn.

PolynomialDivision, 2025-03-03:
Actually, it is constantly sharing probe and beacon information when they arrive and periodically pulling client information (default is 10 seconds), which is configurable with the option `update_client '10'`.

Darin755, 2025-03-04:
Thanks for the reply. I'm slowly working on a management system for OpenWRT that uses gossip and CRDT's to manage a bunch of devices in a decentralized and scalable way. I'm building it to focus on security so it is a little different than DAWN. If I ever get a working version of my project I might look to add band steering.