berlin-open-wireless-lab/DAWN #247: DAWN ubus API will produce invalid JSON and make luci-app-dawn unusable

CoelacanthusHex, 2025-02-28:
Normally, `ubus call dawn get_network` will produce a JSON like
```json
{
  "SSID": {
    "AP_MAC1": { AP_INFO1 },
    "AP_MAC2": { AP_INFO2 },
    "AP_MAC3": { AP_INFO3 }
  }
}
```
But don't know what caused it, on my device, it produces invalid JSON like
```json
{
  {
    "AP_MAC1": { AP_INFO1 }
  },
  "SSID": {
    "AP_MAC2": { AP_INFO2 },
    "AP_MAC3": { AP_INFO3 }
  }
}
```
Some APs were split into another JSON object lacking the key-value pair key, which is not valid in JSON.
In my case, it's
```json
{
        {
                "XX:XX:XX:XX:XX:D8": {
                        "channel": 6,
                        "freq": 2437,
                        "channel_utilization": 4,
                        "num_sta": 0,
                        "ht_support": true,
                        "vht_support": true,
                        "op_class": 81,
                        "local": true,
                        "neighbor_report": "xxxx",
                        "iface": "phy0.0-ap0",
                        "hostname": "router1"
                }
        },
        "My Wi-Fi": {
                "XX:XX:XX:XX:XX:F8": {
                        "channel": 61,
                        "freq": 6255,
                        "channel_utilization": 1,
                        "num_sta": 0,
                        "ht_support": true,
                        "vht_support": true,
                        "op_class": 137,
                        "local": true,
                        "neighbor_report": "xxxx",
                        "iface": "phy0.2-ap0",
                        "hostname": "route1"
                },
                "YY:YY:YY:YY:YY:D2": {
                        "channel": 11,
                        "freq": 2462,
                        "channel_utilization": 36,
                        "num_sta": 0,
                        "ht_support": true,
                        "vht_support": false,
                        "op_class": 81,
                        "local": false,
                        "neighbor_report": "xxxx",
                        "iface": "phy0-ap0",
                        "hostname": "router2"
                },
                "YY:YY:YY:YY:YY:D3": {
                        "channel": 40,
                        "freq": 5200,
                        "channel_utilization": 1,
                        "num_sta": 0,
                        "ht_support": true,
                        "vht_support": true,
                        "op_class": 129,
                        "local": false,
                        "neighbor_report": "xxxx",
                        "iface": "phy1-ap0",
                        "hostname": "router2"
                }
        }
}
```
It makes luci-app-dawn unhappy and throws "No related RPC reply". In browser devtools, we can see luci-app-dawn got an invalid JSON.

I use OpenWrt SNAPSHOT r28914-be181cb3b3 / LuCI Master 25.057.54985~eb422f5 on BananaPi R4 (Router1) and 23.05.03 on 360T7 (Router2).

CoelacanthusHex, 2025-02-28:
It seems only 2.4G AP on Router1 will trigger this issue. Although when the issue was triggered, 5GHz AP may appear in an invalid JSON object as well, but 2.4G AP will always be in an invalid JSON object. And if I disable 2.4G BSSID and restart DAWN to get rid of old data, the issue disappeared. Disabling other APs seems can't workaround the issue.

CoelacanthusHex, 2025-02-28:
> It seems only 2.4G AP on Router1 will trigger this issue. Although when the issue was triggered, 5GHz AP may appear in an invalid JSON object as well, but 2.4G AP will always be in an invalid JSON [...]

It's just that the probability is relatively low. Although I disable 2.4GHz AP, 5GHz AP will appear in an invalid JSON object as well after a longer period.

PolynomialDivision, 2025-03-03:
Do you have just one ssid so everything should be under "My Wi-Fi"?

CoelacanthusHex, 2025-03-04:
> Do you have just one ssid so everything should be under "My Wi-Fi"?

Yes. I just have one SSID.

ivocavalcante, 2025-06-15:
Same here. Restarting DAWN seems to fix issue, surprisingly enough.

s09289728096, 2025-09-24:
I dig deeply and find out the real reason that why json format is broken in my case. The main reason is in hostapd and iwinfo.

I found that dawn uses iwinfo library for getting wireless information. `iw->ssid` will get ssid by different function which depends on different wireless driver. In my case, mt76, it's cfg80211 based driver, so the `iw->ssid` function will call `nl80211_get_ssid` in iwinfo_nl80211.c

Here's the code in iwinfo:

```
static int nl80211_get_ssid(const char *ifname, char *buf)
{
    char *res;
    struct nl80211_ssid_bssid sb = { .ssid = (unsigned char *)buf };

    /* try to find ssid from scan dump results */
    res = nl80211_phy2ifname(ifname);
    sb.ssid[0] = 0;

    nl80211_request(res ? res : ifname, NL80211_CMD_GET_SCAN, NLM_F_DUMP,
                    nl80211_get_ssid_bssid_cb, &sb);

    /* failed, try to find from hostapd info */
    if (sb.ssid[0] == 0)
        nl80211_hostapd_query(ifname, "ssid", sb.ssid,       // <<-- Read string here
                              IWINFO_ESSID_MAX_SIZE + 1);

    /* failed, try to obtain Mesh ID */
    if (sb.ssid[0] == 0)
        iwinfo_ubus_query(res ? res : ifname, "mesh_id",
                          buf, IWINFO_ESSID_MAX_SIZE + 1);

    return (sb.ssid[0] == 0) ? -1 : 0;
}
```

In this function, there are 3 different ways to get ssid:
1. by the request according to phy
2. by keyword "ssid"
3. by keyword "mesh_id"

When we dig deeper, in function `nl80211_hostapd_query`, aka `__nl80211_hostapd_query`:

```
static int __nl80211_hostapd_query(const char *ifname, ...)
{
    struct nl80211_if_data data = {};
    va_list ap, ap_cur;
    char *phy, *search, *dest, *key, *val, buf[128];
    int len, found = 0, match = 1;
    FILE *fp;

    if (nl80211_get_if_data(ifname, &data))
        return 0;

    if (data.mode != IWINFO_OPMODE_MASTER && data.mode != IWINFO_OPMODE_AP_VLAN)
        return 0;

    phy = nl80211_ifname2phy(ifname);

    if (!phy)
        return 0;

    if (data.radio_idx >= 0)
        snprintf(buf, sizeof(buf), "/var/run/hostapd-%s.%d.conf", phy, data.radio_idx);
    else
        snprintf(buf, sizeof(buf), "/var/run/hostapd-%s.conf", phy);
    fp = fopen(buf, "r");

    if (!fp)
        return 0;

    va_start(ap, ifname);

    /* clear all destination buffers */
    va_copy(ap_cur, ap);

    while ((search = va_arg(ap_cur, char *)) != NULL)
    {
        dest = va_arg(ap_cur, char *);

......
```
This function will try to get hostapd configuration in `/var/run/hostapd-phyX.Y.conf`
In my case, becase I use new wifi-scripts, which uses ssid2 instead of ssid, this leads to DAWN get empty strings in ssid and breaks json format when trys to create table in json by `blobmsg_open_table(&buf, ssid)`
https://github.com/openwrt/openwrt/blob/4eae48d9dcd374a2f5cac5247bc0caeb1e49e7d1/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/ap.uc#L47

For you guys to confirm this root cause, the simple way is **edit every hostapd.conf in /var/run/ directory to add `ssid` parameter into** and execute `/etc/init.d/dawn restart`. After this, luci-app-dawn will render correctly.

[image]

s09289728096, 2025-09-24:
It seems iwinfo seldoms accepts PR on github... I'll try for weeks to check if there any issue, and open a PR for it.